ARM-software/CMSIS-NN #13: Compilation fail with arm-gcc-none-eabi 12.2.0

asimov-aiz, 2022-11-15:
I using the arm gcc from official website: https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads

```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (Arm GNU Toolchain 12.2.MPACBTI-Bet1 (Build arm-12-mpacbti.16)) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

When I try to compile with steps from: https://github.com/ARM-software/CMSIS-NN#building-cmsis-nn-as-a-library

I got the compilation error:

```
[ 60%] Building C object CMakeFiles/cmsis-nn.dir/Source/SoftmaxFunctions/arm_softmax_s8.c.obj
during RTL pass: combine
/CMSIS-NN/Source/SoftmaxFunctions/arm_softmax_s8.c: In function 'arm_exp_on_negative_values_mve_32x4':
/CMSIS-NN/Source/SoftmaxFunctions/arm_softmax_s8.c:74:1: internal compiler error: in trunc_int_for_mode, at explow.cc:59
   74 | }
      | ^
0x7f2a15836d8f __libc_start_call_main
        ../sysdeps/nptl/libc_start_call_main.h:58
0x7f2a15836e3f __libc_start_main_impl
        ../csu/libc-start.c:392
Please submit a full bug report, with preprocessed source (by using -freport-bug).
Please include the complete backtrace with any bug report.
See <https://bugs.linaro.org/> for instructions.
make[2]: *** [CMakeFiles/cmsis-nn.dir/build.make:580: CMakeFiles/cmsis-nn.dir/Source/SoftmaxFunctions/arm_softmax_s8.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:295: CMakeFiles/cmsis-nn.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

Seems like the issue with mve support. I will need to use latest gcc as my target is also m85 that previous gcc doesn't support.

felix-johnny, 2022-11-15:
@asimov-aiz Thanks for reporting this. We'll have  a look.

felix-johnny, 2022-11-16:
@asimov-aiz It does look like an error in the compiler. I've raised an internal ticket to the tools team on it for now. I'll update here when there is progress on that. 

As a way forward, I see two options.
1. CMSIS-NN disables MVE optimizations for softmax_s8 or
2. Avoid the beta release of 12.2.0 and use production version 11.3 and build for Cortex-M55 instead. You can use the binary/library from that for your project. 

Since 12.2.0 is not a production release, my suggestion is to go with option 2. If you have access to armclang on the other hand, you can use that with cortex-m85 as the cpu option.

felix-johnny, 2022-12-06:
Link to fix: https://gcc.gnu.org/pipermail/gcc-patches/2022-December/607963.html

felix-johnny, 2022-12-23:
@asimov-aiz .. The workaround for this is now delivered in CMSIS-NN. The fix in the toolchain is however planned for the next release and is not part of the released toolchain on Dec 22.  With this, I'll close the ticket and do get back for any clarifications regarding this or anything else!

/Felix

closed 2022-12-23 by felix-johnny