ARM-software/CMSIS-NN #19: Potential bug in  arm_depthwise_conv_wrapper_s8_get_buffer_size

SebastianBoblest, 2022-11-18:
We think there is a bug in arm_depthwise_conv_wrapper_s8_get_buffer_size for cases, 
where arm_depthwise_conv_wrapper_s8 calls arm_depthwise_conv_3x3_s8. 
arm_depthwise_conv_3x3_s8 has a buffer size of zero, but the arm_depthwise_conv_wrapper_s8_get_buffer_size function 
does not handle the case 
(filter_dims->w == 3) && (filter_dims->h == 3) && (dw_conv_params->padding.h <= 1) && (dw_conv_params->padding.w <= 1)) special.
It should return 0 then.
Please compare 
https://github.com/ARM-software/CMSIS-NN/blob/ca5dc34313be2ee5c46652917c30baac96c52621/Source/ConvolutionFunctions/arm_depthwise_conv_wrapper_s8.c#L66

and

https://github.com/ARM-software/CMSIS-NN/blob/ca5dc34313be2ee5c46652917c30baac96c52621/Source/ConvolutionFunctions/arm_depthwise_conv_wrapper_s8.c#L124

@ulrikhjort @vdkhoi @bewagner

felix-johnny, 2022-11-21:
@SebastianBoblest  Thanks for reporting this. You are right that the 3x3 case which is for processors without Helium Technology(Armv8.1M) should not require an additional memory. Ideally the wrapper_get_buffer() function should have the same condition checks as the wrapper() function. 
   As for the consequence of this, there will be an additional memory for scratch buffer that is requested and this is not a buffer overflow case. 
 The additional memory requested(that isn't needed) is for two int8 im2col buffers. If my understanding is right memory allocators do not add up the memory requires scratch buffers for each layers but rather reuse them between layers. So, that is a worst case of ker_w * ker_h * input_ch * 2 bytes that is over allocated.

 We'll provide a fix for this. The fix is not part of the 4.0.0 release.

felix-johnny, 2022-12-06:
Fix is merged.

closed 2022-12-06 by felix-johnny