ARM-software/CMSIS-NN #47: How to implement a neural network by using CMSIS-NN？

Raise-me-up, 2023-03-13:
Is there any tutorial?  I am confused about the cmsis-nn transform of nn.lstm, nn.layernorm, nn.prelu etc。

felix-johnny, 2023-03-13:
@Raise-me-up thanks for the question. CMSIS-NN is an optimization library that provides optimization for some of the ML operators in TensorFlow Lite for Microcontrollers framework. 

https://github.com/ARM-software/CMSIS-NN/tree/main/Examples has some examples to see how one can get started on it.

Raise-me-up, 2023-03-13:
@felix-johnny Thanks for your reply! I have read your example. However, it is based on Tensorflow lite micro library, and I just want to implement a neural network by using cmsis-nn purely, not the third party library. Are there any other examples?

Raise-me-up, 2023-03-14:
@felix-johnny OK! Thanks for your detailed reply!

closed 2023-03-14 by Raise-me-up