ARM-software/CMSIS-NN #53: Requantize INT_32 to INT8 using arm_nn_requantize

Theoo1997, 2023-03-24:
Hi, 
I am using [this](https://github.com/ARM-software/CMSIS-NN/blob/main/Source/FullyConnectedFunctions/arm_fully_connected_s8.c) FC s8 CMSIS layer for a project. After we invoke  a FC s8 kernel we have Int32 output that we need to dequantize-requantize (Int32->FP->Int8).  For example if we have this FC layer in the image to dequantize-requantize the 32-bit output of the FC layer we need to apply this equation: Output_Int8 = (0.0039215669967234135*Weight_mult/0.14308570325374603) *Output_Int32 + 69. => Multiplier * Output_Int32 + shifter
[image]

CMSIS-NN has arm_nn_requantizefunction that takes as input Int32 Multiplier and shifter and returns (val * multiplier)/(2 ^ shift). Can you explain me how to map the above FP Multiplier and Int32 shifter into the requantize function?

felix-johnny, 2023-03-24:
@Theoo1997 Tanks for the question!
 I am assuming this is a tflite network. If you use the TFLM framework then multiplier/shift  is calculated by the framework. Is there a reason for not using the framework? It is quite some overhead for the non-framework approach. 
 If for some reason you do not want to use the framework, then the closest I can point you to is https://github.com/ARM-software/CMSIS-NN/blob/main/Tests/UnitTest/generate_test_data.py#L843 .. Or look in the code base of https://github.com/tensorflow/tflite-micro ..

Hope that helps and let me know if you have any follow ups..

Felix

Theoo1997, 2023-03-24:
> @Theoo1997 Tanks for the question! I am assuming this is a tflite network. If you use the TFLM framework then multiplier/shift is calculated by the framework. Is there a reason for not using the [...]

Thank you vary much for you answer.

closed 2023-03-24 by Theoo1997