ARM-software/CMSIS-NN #58: Create dynamic library for Cortex-M33

MiguelCosta94, 2023-04-20:
Hello,

Is it possible to compile a dynamic library for the Cortex-M33? I changed the CMakeLists.txt file as detailed below but I still get a static library.

`cmake_minimum_required(VERSION 3.15.6)
set(CMAKE_POSITION_INDEPENDENT_CODE ON)
set(BUILD_SHARED_LIBS ON)
project(CMSISNN)
add_library(cmsis-nn SHARED)
target_compile_options(cmsis-nn PRIVATE -Ofast -fPIC)
target_include_directories(cmsis-nn PUBLIC "Include")
add_subdirectory(Source)
set_target_properties(cmsis-nn PROPERTIES
    POSITION_INDEPENDENT_CODE ON
)`

mansnils, 2023-04-24:
Hi @MiguelCosta94 
CMSIS-NN dynamic library is not officially supported. As far as I know nobody has done it. That being said, it should be possible.

Zheng-Bicheng, 2023-06-15:
What are the requirements for compiling CMSIS-NN into a dynamic library? I am curious where CMSIS-NN in the form of dynamic libraries can be used.

mansnils, 2023-07-03:
I don't think there are any specific CMSIS-NN requirements for this. It is a generic CMake/build/compiler issue/question on how to build a dynamic library.