ARM-software/CMSIS-NN #70: Is there a roadmap?

kit-high, 2023-09-21:
Hi, thank you for your work!

Is there a future roadmap in long/short term  like adding  RNN... transformer...?

mansnils, 2023-09-29:
Hi @kit-high, thanks for your interest in CMSIS-NN.
In general we add optimizations for operators that are supported by TFLM as we are an optimization library and not a ML framework. You can expect new operators like transpose conv and new data types like int4.
Is there any particular operator you are interesting in?

kit-high, 2023-10-13:
@mansnils 
I apologize for the delayed response.

I agree with you that it's not an ML framework.
However, since it already has implementations for somewhat complex ML methods like LSTM, I was expecting that it would also implement methods that are not yet included, such as RNNs or Transformers.

As an engineer, I feel that the advancement of Light Edge AI is closely tied to the development of this library. If the opportunity arises, I would like to contribute by making a pull request. Thank you.

closed 2023-10-13 by kit-high

mansnils, 2023-10-17:
Thanks @kit-high. Your contribution is welcome. Anyhow a potential new operator should have an equivalent reference kernel in TFL/TFLM.