ARM-software/CMSIS-NN #87: Occur an ASM error when compiling arm_nn_mat_mul_core_4x_s8

WORsjlt, 2023-11-21:
Processor: Cortex-M55
Error message: 
cmsis-nn/Source/NNSupportFunctions/arm_nn_mat_mul_core_4x_s8.c: In function 'arm_nn_mat_mul_core_4x_s8':
/(my project path)/cmsis-nn/Include/Internal/arm_nn_compiler.h:97:23: error: 'asm' operand has impossible constraints
   97 |         #define __ASM __asm
      |                       ^~~~~
cmsis-nn/Source/NNSupportFunctions/arm_nn_mat_mul_core_4x_s8.c:84:9: note: in expansion of macro '__ASM'
   84 |         __ASM volatile(" .p2align 2                             \n"
      |         ^~~~~

Why is this happening? And how to fix it?
Appreciate!

mansnils, 2023-11-21:
Hi @WORsjlt 
Thanks for reporting this. What compiler and version are you using?

mansnils, 2023-11-21:
Also would like to check if it may be related to compiler options?
https://github.com/ARM-software/CMSIS-NN#compiler-options

WORsjlt, 2023-11-22:
Hi @mansnils 
Here is the output message about compiler and version when compiling:
GNU C17 (Arm GNU Toolchain 11.3.Rel1) version 11.3.1 20220712 (arm-none-eabi)
	compiled by GNU C version 4.8.5 20150623 (Red Hat 4.8.5-44), GMP version 6.2.1, MPFR version 3.1.6, MPC version 1.0.3, isl version isl-0.15-1-g835ea3a-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
Compiler executable checksum: 801c9e4705ed9ec92526a3ce66f3ab92
GNU assembler version 2.38 (arm-none-eabi) using BFD version (Arm GNU Toolchain 11.3.Rel1) 2.38.20220708

Hope it helps!

mansnils, 2023-11-22:
It looks like a supported compiler. Question is how to reproduce this. Do you build like described here: https://github.com/ARM-software/CMSIS-NN#building-cmsis-nn-as-a-library
Or do you set some custom build options?

WORsjlt, 2023-11-22:
I built cmsis-nn inside a project using Nuttx. I defined -DTARGET_CPU=cortex-m55 and some custom options, but I'm not sure which option produces this. Here are some options filtering out header files:
-imultilib thumb/v8-m.main+dp/hard -iprefix /(my project path)/prebuilts/gcc/linux/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/ -isysroot /(my project path)/prebuilts/gcc/linux/arm/bin/../arm-none-eabi -D__USES_INITFINI__ -D __NuttX__ -isystem /(my project path)/nuttx/include cmsis-nn/Source/NNSupportFunctions/arm_nn_mat_mul_core_4x_s8.c -quiet -dumpdir cmsis-nn/Source/NNSupportFunctions/ -dumpbase arm_nn_mat_mul_core_4x_s8.c.(my project path).cmsis-nn.c -dumpbase-ext .c -mlittle-endian -mtune=cortex-m55 -mfpu=fpv5-d16 -mfloat-abi=hard -mthumb -mlibarch=armv8.1-m.main+fp.dp+mve.fp -march=armv8.1-m.main+fp.dp+mve.fp
-EL -march=armv8.1-m.main+fp.dp+mve.fp -mfloat-abi=hard -mfpu=fpv5-d16 -meabi=5 -mthumb -mimplicit-it=always

felix-johnny, 2023-11-22:
@WORsjlt The optimization level is missing in this. Could you try with atleast -O1 or higher?

WORsjlt, 2023-11-22:
@felix-johnny I used -O3.

mansnils, 2023-11-22:
Can you try without `-fno-omit-frame-pointer`?

WORsjlt, 2023-11-23:
Cancelling` -fno-omit-frame-pointer` works! Thank you very much! @mansnils @felix-johnny