ARM-software/CMSIS-NN #89: reserved identifier violation

elfring, 2023-12-02:
I would like to point out that identifiers like “[`_ARM_NNFUNCTIONS_H`](https://github.com/ARM-software/CMSIS-NN/blob/bfc54edb61e873039ec0857cacc40df36b1d644e/Include/arm_nnfunctions.h#L112-L113 "Update candidate")” and “[`_UART_STDOUT_H_`](https://github.com/ARM-software/CMSIS-NN/blob/040da18234b597867bd54d736ca4ceae545b2853/Tests/UnitTest/Corstone-300/uart.h#L19-L20 "Another update candidate")” [do not fit](https://wiki.sei.cmu.edu/confluence/display/cplusplus/DCL51-CPP.+Do+not+declare+or+define+a+reserved+identifier#DCL51CPP.Donotdeclareordefineareservedidentifier-NoncompliantCodeExample%28HeaderGuard%29 "Do not declare an identifier which is reserved for the compiler implementation.") to the expected naming convention of the C++ language standard.
Would you like to adjust your selection for unique names?

mansnils, 2023-12-04:
Thanks for reporting this @elfring. Yes we would like to update this.

mansnils, 2023-12-05:
Yes that might be useful. Maybe a clang-tidy config file in the root of the repo similar to: https://github.com/ARM-software/CMSIS-NN/blob/main/.clang-format. Do you have any suggestions?

elfring, 2023-12-06:
:crystal_ball: Under which circumstances are you going to rename affected include guards (for example)?

mansnils, 2023-12-18:
Closing as issue is solved. @elfring - feel free to open a new issue if you have any proposal for a development tool like clang-tidy.

closed 2023-12-18 by mansnils

elfring, 2023-12-18:
Thanks for [your adjustment of affected include guards](https://github.com/ARM-software/CMSIS-NN/commit/29c331e2f86e2cb131fdaee57954dd475ac528d1 "Rename include guards to align to naming convention").

:crystal_ball: Can it influence any changes for related software components?