ARM-software/CMSIS-NN #92: Is there some easy to get started guide?

abdul-rehman-2050, 2023-12-15:
I wanted to use this but don't know where to get started. I want to get aware with the work flow of how CMSIS NN work, with some easy to use example like some sine wave predictor which TensorFlow lite library comes with Arduino etc. 
Is there getting's tarted guide for this? 
Do I have to train the model on the device or I can convert the TensorFlow trained and quantized model with this? Can I run some simple NN example on STM32F103C8 blue pill board? no matter how small the example is. I just want to get use to with the work flow. 
Thanks.

mansnils, 2023-12-18:
Hi @abdul-rehman-2050 
Thanks for your interest in CMSIS-NN. I would recommend starting with Tensorflow Lite for MicroControllers as that is used for deploying and CMSIS-NN is just a library. TFLM has support for bluepill and I think Arduino as well. You don't have to train the model on device. You can use TfliteConverter to quantize your model and get a tflite binary that can be run with TFLM. In the TFLM repo there are also many examples to get started with and you can run them with CMSIS-NN optimized kernels. Here are some links.
https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/docs/arm.md
https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/examples
https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/kernels/cmsis_nn/README.md
https://www.tensorflow.org/api_docs/python/tf/lite/TFLiteConverter

thorinf, 2024-09-15:
https://github.com/ARM-software/CMSIS-NN/issues/145

I've made a similar request here. Would be great to see the most barebones example possible within the example folder of this repo.

mansnils, 2024-09-19:
Closing this. See further comments in duplicate issue: https://github.com/ARM-software/CMSIS-NN/issues/145

closed 2024-09-19 by mansnils