ARM-software/CMSIS-NN #101: missing arm_s8_to_s16_unordered_with_offset.c when I included the pack

tonywu-uec, 2024-02-05:
I try to incorporate the CMSIS-NN into my project. However, I saw error message:

.\objects\myProject.axf: Error: L6218E: Undefined symbol arm_s8_to_s16_unordered_with_offset (referred from arm_convolve_s8.o).

I traced the function in the CMSIS-NN and found out the source code arm_s8_to_s16_unordered_with_offset.c is in the
directory C:\Users\XXX\AppData\Local\Arm\Packs\ARM\CMSIS-NN\4.1.0\Source\NNSupportFunctions. But the pack didn't
include the file.

[image]
[image]
[image]

Did the pack miss it accidentally? How can I add it back into the pack?

Best Regards,

/Tony

mansnils, 2024-02-05:
Hi @tonywu-uec 
Unfortunately there was a mistake in the 4.1.0 pack/component, sorry about that. Perhaps this will work?
- Rename the pack to a zip file
- Unzip and add  <file category="source" name="Source/NNSupportFunctions/arm_s8_to_s16_unordered_with_offset.c"/> to ARM.CMSIS-NN.pdsc
- Rezip and rename to original name
Or if you could use the 5.0.0 CMSIS-NN pack/component instead?

iga31, 2024-02-27:
Hi @mansnils 
`arm_nn_mat_mult_kernel_row_offset_s8_s16.c` and `arm_nn_vec_mat_mul_result_acc_s8_s16.c`
are not included in current `pdsc`.

mansnils, 2024-02-27:
Thanks @iga31. We will will fix this.

closed 2024-04-02 by mansnils