ARM-software/CMSIS-NN #127: Move test platform to other FVPs

njsyw1997, 2024-04-03:
Hi, 

I am trying to figure out how to use FVP as a test platform. I successfully build tests as the tutorial shows. But when I try to switch to Corstone SSE-310 FVP to test how my code performs on M85, it fails. It seems code in [Corstone-300](https://github.com/ARM-software/CMSIS-NN/tree/main/Tests/UnitTest/Corstone-300) is hardware-specific. How can I generate these codes for other FVPs?

Thanks!

mansnils, 2024-04-04:
Hi @njsyw1997 
So I guess your question is about running the CMSIS-NN unit tests. Please note you can note you can not use the Arm(R) Corstone(TM)-300 FVP for performance testing as explained [here](https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/benchmarks#run-on-fvp-based-on-arm-corstone-300-software).
Yes the FVP emulates a[Arm(R) Cortex(R)-M55 system, but it is backwards
compatible. so we use it in the unit test to run code compiled for e.g. a Cortex-M7. So please use M55 instead of M85 (although it might still work), and from CMSIS-NN unit test perspective the same code will be run.
So why it does not work in your case is hard to say especially without any logs.
You can try to run this [script](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/tools/ci_build/test_cortex_m_corstone_300.sh) which download every dependency from scratch and should work for out of the box. The CMSIS-NN unit tests use the FVP in the same way.

njsyw1997, 2024-04-04:
@mansnils I see. I just found the directory [corstone-310](https://review.mlplatform.org/plugins/gitiles/ml/ethos-u/ethos-u-core-platform/+/refs/tags/24.02/targets/corstone-310/) from the script. Link cmake to it and now the test works.

Since FVP cannot measure performance, is there any other way to measure performance such as cycles and memory footprint except the real hardware?  

Thank you very much!

mansnils, 2024-04-05:
Cycles can be measured as examplified [here](https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/benchmarks) or [here](https://review.mlplatform.org/plugins/gitiles/ml/ethos-u/ethos-u-core-software/+/refs/heads/master/applications/inference_process/src/inference_process.cpp#154). Just note that for CPU measurements (i.e. CMSIS-NN) to get valid measurements it has to be run on real HW.
I can also point you to this unit test [script](https://github.com/ARM-software/CMSIS-NN/blob/main/Tests/UnitTest/unittest_targets.py) (not doing cycle measurements) that should be plug and play on STM32 Nucleo boards.

njsyw1997, 2024-04-09:
I just met some small bugs in FVP setup. I tried to use the code from /ethos-u-core-platform/targets/corstone-300 to establish the tests but the function `printf` do not perform correctly. I made a minimal example in this github [repo](https://github.com/njsyw1997/avh_example). Could you please have a look?

Thank you very much!

mansnils, 2024-04-10:
I am not sure what you are trying to fix. I can assure you printf works when running the unit test in its current setup. This raises however the question of having a CI script for running the unit tests.

njsyw1997, 2024-04-12:
I just copy the code directly from ethos-u-core-platform but it does not work. I am just wondering if I use FVP wrong. Since this is CMSIS-NN repo, it might not be a good place to discuss it here. I will close the issue.

Thanks for your patience!

closed 2024-04-12 by njsyw1997