ARM-software/CMSIS-NN #141: Why does convolve_s8 use an int16 buffer?

tomhepworth, 2024-08-07:
I am trying to understand the function mat_mult_kernel_s8_16. It is called from the function arm_convolve_s8 where I can see that input_data and filter_data are 8-bit. What conditions cause mat_mult_kernel_s8_16 to be used? What is in the ctx->buf that gets passed to it?

I understand i8 * i8 --> int16, but if the output type is 8-bit then I do not understand why it is not rescaled back to 8bit before being stored. Is it something to do with the im2col operation? If so then my understanding was that im2col was essentially rearranging the input/filter. Why should that change the type?

felix-johnny, 2024-08-07:
@tomhepworth Thanks for the question! The function you refer to https://github.com/ARM-software/CMSIS-NN/blob/main/Source/NNSupportFunctions/arm_s8_to_s16_unordered_with_offset.c called from https://github.com/ARM-software/CMSIS-NN/blob/9d924bdaee51ca8e0c4e86779bbb6d0c9644e555/Source/ConvolutionFunctions/arm_convolve_s8.c#L190 is an im2col step. 
    We need int16 there as the input is (8 bit input + 8 bit input offset). This is then used in the MAC operation like in https://github.com/ARM-software/CMSIS-NN/blob/9d924bdaee51ca8e0c4e86779bbb6d0c9644e555/Source/ConvolutionFunctions/arm_convolve_s8.c#L314 .

Hope that helps..

tomhepworth, 2024-08-07:
Thank you for the fast reply.

Could you please explain what this offset is? Could it be the quantization zero point as in other fixed point neural network operations?

felix-johnny, 2024-08-07:
Correct.. It is the zero point.

closed 2024-08-07 by tomhepworth