ARM-software/CMSIS-NN #145: Request: Simple Examples

thorinf, 2024-09-15:
Hi,

Are there any very simple examples available? Image Recognition is great, but I'm thinking even simpler.

For example, can I take a model with a single FC layer, just a single weight and bias, and compile it for the cortex-m via tflite?

Thanks

ArmRyan, 2024-09-16:
Hey @thorinf, It sounds like the easiest way for you to get started would be to look at some of the tflite-micro examples https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/examples 

Cmsis-nn can be built as a target for tflite-micro and would be the main way to use the library with your tflite files. If you can get the basic use of tflite-micro down then you can you use any small single layer models the same way and run it through cmsis-nn.

thorinf, 2024-09-16:
Hi,

Thanks for getting back to me so fast.

```shell
make -f tensorflow/lite/micro/tools/make/Makefile TARGET=cortex_m_generic TARGET_ARCH=cortex-m7 microlite
make -f tensorflow/lite/micro/tools/make/Makefile TARGET=cortex_m_generic TARGET_ARCH=cortex-m7 OPTIMIZED_KERNEL_DIR=cmsis_nn microlite
```

I did manage to get these to build, just running into difficulties building my project. VSCode can find the tflite-micro headers when i click on them but my build process can't. I did put the libs in the Makefile, but I'm probably making a mistake - I should be able to fix it.

ArmRyan, 2024-09-16:
Great! Glad to see you making some progress.

The issue with adding examples to cmsis-nn is that the library it not really intended to be used on its own, so its often better to use the tflite micro examples instead. You can directly call the functions, but you would need to have your own framework built around it since we do not do any memory management or handle persistent data between functions etc.