ARM-software/CMSIS-NN #150: Question/Clarification Request: Test-Data for Addition

MercurCodes, 2024-10-16:
Hi there,

I'm currently working with some code related to the add function  from the basic math and during testing we ran into some confusion. I want to utilize the Test Data from add and was confused about the configuration.
If one looks at input_1 and input_2, one sees that -92 and 72 is given. Adding those numbers up results in -20, however, the expected data is -10, which aligns with result if one utilizes the given configuration. 
However, if the configuration of ADD_OUTPUT_SHIFT would be -20 instead of -19, the addition would return "cleaner" values (-10 instead of -20). 
Was this done by design? 
What was the idea behind this?

Thank you for sharing your insights with me.

ArmRyan, 2024-10-17:
Hey, I am not sure if I fully understand the question so would you be able to explain a bit more by what you mean by "cleaner" values and maybe if this is related to a certain hardware implementation or just a generic question about how it works?

I can give a brief overview for now though!

So currently cmsis-nn aims to be bit exact with the reference kernels from tflite-micro as the main use case is to use cmsis-nn as a library of optimized kernels through tflite-micro. So to maintain bit-exactness we follow the [tflite quantization specification](https://ai.google.dev/edge/litert/models/quantization_spec). It says litert but it used to be tflite! This scheme dictates how quantized data should be described. The data or configuration in question is essentially auto generated by creating a single layer keras model with randomized input data and then quantized through the tflite converter to create a tflite model. We then extract the parameters including the quantization parameters such as the shift and offsets and write them into our test data as either .c or .h files.

Here is the reference op for tflite that we aim to optimize: https://github.com/tensorflow/tflite-micro/blob/e86d97b6237f88ab5925c0b41e3e3589a1560d86/tensorflow/lite/kernels/internal/reference/integer_ops/add.h#L180

I hope this helps.

MercurCodes, 2024-10-18:
Hey,
Thanks for the answer. Your resources were definitely already a big help. 
I just want to expand a little bit on my question and clarify it a little. 
If you look at the tests aimed at int8 elementwise add the first value pair is -92 and 72. If you add those two, you receive -20, which we expected as well. However, due to the setup of the configuration of the tests, the result is -10, which we get, is also correct.
We can see, that ( -92 * 2^30 / 2^11) + (72 * 2^30 / 2^11) = -92 * 2^(30 - 11) + 72 * 2^(30 - 11)  = (-92 + 72) * 2^19 = -20 * 2^19
Which then gets further processed: -20 * 2^19 * 2^30  / 2^50  = -20 * 2^(19 + 30 - 50) = -20 * 2^-1 = -10
Explanation regarding the multiplications:
Input values (72, -92)
- * 2^30 -> Scale = 1073741824 = 2^30 ()
- * 2^11 -> Shift to the right by: 31 - 20 (ADD_LEFT_SHIFT) -1 + 1 (Rounding)  

Output value:
- * 2^30 -> Scale = 1073741824 = 2^30 ()
- *2^50 -> Shift to the right by: 31 - (-19) (ADD_OUTPUT_SHIFT) - 1 + 1 (Rounding) = 50

In our testing we adapted your configurations without looking at the reference values and expected the output of -92 + 72 = -20, but it was instead -10
We get, that this is the correct solution as shown above, however, we were wondering why this decision was made. Why didn't you test with ADD_OUTPUT_SHIFT = -20. This would have resulted in -20 instead of the -10 which is now expected. 
We are currently utilizing this test data in the context of a Master Thesis and we would like to explain in the thesis why this additional shift is in the test data. We want to show proper scientific work was done and think you had an idea behind this test setup. A official source on this would look really good in the thesis instead of just speculating why this is done this way.

mansnils, 2024-10-18:
Hi @MercurCodes 
As Ryan said CMSIS-NN provides bit-exactness to TFL/TFLM and we [generate test data](https://github.com/ARM-software/CMSIS-NN/blob/main/Tests/UnitTest/README.md#refactoring-of-generate_test_datapy) for the unit tests that align to this requirement. So the inputs that we "decided on" for the test case that contains ADD_OUTPUT_SHIFT are here: https://github.com/ARM-software/CMSIS-NN/blob/5f8f1a96797cfce64032492151b01cf0e1c97f06/Tests/UnitTest/generate_test_data.py#L3080
The ADD_OUTPUT_SHIFT is then set here: https://github.com/ARM-software/CMSIS-NN/blob/5f8f1a96797cfce64032492151b01cf0e1c97f06/Tests/UnitTest/add_mul_settings.py#L124
So we didn't really decide to set ADD_OUTPUT_SHIFT as anything else than -19 would break the requirement.
To understand how ADD_OUTPUT_SHIFT is derived it may help to regenerate the test data for the test case named 'add' that contains ADD_OUTPUT_SHIFT.

MercurCodes, 2024-10-18:
Hei @mansnils 
Thanks for the quick answer. I now understand how the values were generated and understand the values.
I'm sorry I missed the ReadMe. Thanks for pointing out and linking to the relevant functions.

closed 2024-10-18 by MercurCodes