ARM-software/CMSIS-NN #161: Stack Overflow Occurs When Calling arm_convolve_s8 Function

Ijustakid, 2025-01-02:
I attempted to execute a convolutional neural network on a Cortex-M55, where the input size is 1x3x256x144 (NCHW), the kernel size is 12x3x3x3, padding is (1, 1), and stride is (1, 1). The output size is 1x12x256x144 (NCHW).

The compilation command executes correctly, and the compilation .bin are successfully generated.

However, during runtime, a stack overflow crash occurs.

When I reduce the input size to 1x1x16x16 while keeping the kernel size unchanged, the crash does not occur.

Could it be that the arm_convolve_s8 function's recursive depth is too deep when dealing with large feature maps, causing the stack overflow?

Do you have any relevant experience with this issue?

BRs,
Thanks!

Ijustakid, 2025-01-24:
> Hi [@Ijustakid](https://github.com/Ijustakid) , I have tried to reproduce this and failed. Could you provide more input or a sample model that you see this failure in? Perhaps the model is just too large for the device that you are running on or you are allocating memory incorrectly on your platform?

Yes, it is indeed my problem, my device RAM doesn't have that much space to allocate variables. Thanks for your reply.

I have another question, is the data format supported by CMSIS_NN NCHW or NHWC?
For example, for conv's inputs and weights which are 2D structure data, how to arrange them into 1D array variables?

Thanks!

BRs

ArmRyan, 2025-01-24:
The data format should be NHWC just like tflite micro's default format! We don't currently support any NCHW kernels.

So if it helps, when we create rows for a gemm kernel in a convolution we would do something similar to
For H
  For W
    memcpy(length(channels))

Source: https://github.com/ARM-software/CMSIS-NN/blob/main/Source/ConvolutionFunctions/arm_convolve_s4.c#L113-L128

OmnishamblesOUO, 2025-02-05:
> The data format should be NHWC just like tflite micro's default format! We don't currently support any NCHW kernels.
> 
> So if it helps, when we create rows for a gemm kernel in a convolution we would do something similar to For H For W memcpy(length(channels))
> 
> Source: https://github.com/ARM-software/CMSIS-NN/blob/main/Source/ConvolutionFunctions/arm_convolve_s4.c#L113-L128
> 
> Is it safe to assume you are not using tflite micro if the data format is NCHW? cmsis-nn is designed to be used alongside tflm. So the easiest way to access the library would be to build it through tflite micro. It should also fail to compile before it gets to the point where it could stack overflow if there is insufficient memory.

Thank you very much! I know that CMSIS-NN uses the NHWC format, which is very helpful for me!

Additionally, yes, I am not using TFLite-Micro.
My model structure is quite simple, so I plan to use CMSIS-NN operators, manually filling in weights, biases, and other parameters to construct a model that can be inferred on Cortex-M55.

However, I still have some questions:

1. I found a script in the project: CMSIS-NN/Tests/UnitTest/model_extractor.py. It seems to be able to extract weights, biases, and some quantization parameters from a TFLite model. However, when I tried to execute it, I encountered the following error:
`File "C:\project\CMSIS-NN\Tests\UnitTest\conv_settings.py", line 113, in __init__
    self.filter_ch = in_ch // groups
ZeroDivisionError: integer division or modulo by zero`

I checked the test_settings.py script. Could you please confirm whether only operators with predefined parameters (such as dimensions) can be extracted?

2. I quantized my model using TFLite, then used
`tensor_data = interpreter.get_tensor(detail['index'])`
to obtain the weights for each layer and planned to fill them into CMSIS-NN.

How should I set some quantization parameters for the convolution operator? For example:
`#define CONV_2_OUT_ACTIVATION_MIN -128
#define CONV_2_OUT_ACTIVATION_MAX 127
#define CONV_2_INPUT_OFFSET 0
#define CONV_2_OUTPUT_OFFSET 0`
Also, for the multiple and shift variables:
`conv_2_output_mult
conv_2_output_shift`
What values should they take?

I really appreciate your help!

BRs.

OmnishamblesOUO, 2025-03-28:
> The data format should be NHWC just like tflite micro's default format! We don't currently support any NCHW kernels.
>
> So if it helps, when we create rows for a gemm kernel in a convolution we would [...]

Thank you for your response. After considering the options, I ultimately decided to go with TFLite, and it has indeed helped me avoid these issues. I appreciate your time and assistance!

Thanks!

OmnishamblesOUO, 2025-03-28:
> Hi [@OmnishamblesOUO](https://github.com/OmnishamblesOUO) and thanks for your interest in CMSIS-NN. Just to iterate what perhaps has already been said and that is that CMSIS-NN is a library an is intended to be used by a framework, i.e. TFLite Micro. That being said it is not impossible to use it standalone. In fact the model_extractor.py script was used in such trials and the outcome of that was these were DS_CNN int8 models are run standalone: https://github.com/ARM-software/CMSIS-NN/tree/main/Tests/UnitTest/TestCases/test_arm_ds_cnn_l_s8 https://github.com/ARM-software/CMSIS-NN/tree/main/Tests/UnitTest/TestCases/test_arm_ds_cnn_s_s8 Those may give some direction. The values you are asking about are model specific, so changing the values would give different output of a given value. So not really really a right and wrong answer there unless you want to match the output with some reference.

Thank you for your response. After considering the options, I ultimately decided to go with TFLite, and it has indeed helped me avoid these issues. I appreciate your time and assistance!

Thanks!