ARM-software/CMSIS-NN #164: Usage of buffer created using arm_fully_connected_s8_get_buffer_size

praveen0447, 2025-01-29:
Hi , 

I am trying to use the arm_fully_connected_s8 which needs the buffer created using  arm_fully_connected_s8_get_buffer_size . If I look at the internal implementation  of usage , it is used for setting the accumulator before doing the matrix multiplication as shown below.

            int32_t acc_0 = *kernel_sum++;
            int32_t acc_1 = *kernel_sum++;
            int32_t acc_2 = *kernel_sum++;
            int32_t acc_3 = *kernel_sum++;

Can't we assign zeros here? or am I missing something.

Regards,
Praveen.

felix-johnny, 2025-01-29:
Hi @praveen0447 .. Thanks for the question.. The buffer is expected to be filled with the reduction sum(sum of kernels) and that is used as an optimization to not process the input offset inside the core loop. If we didn't have that optimization, then it would have been initialized with zeros or the bias values. 

Here is the prelude to the arm_fully_connected_s8_get_buffer_size() call where the reduction sum is calculated.

https://github.com/tensorflow/tflite-micro/blob/9b79b9faf208fddb509a0efc671bf338b5675ab9/tensorflow/lite/micro/kernels/cmsis_nn/fully_connected.cc#L354C25-L355C1

You can read about this step in https://github.com/google/gemmlowp/blob/master/doc/low-precision.md#efficient-handling-of-offsets

praveen0447, 2025-01-30:
Thanks for the quick response. It is clarified now..

felix-johnny, 2025-01-30:
Thanks I'll close it then.

closed 2025-01-30 by felix-johnny