ARM-software/CMSIS-NN #165: Understanding Data Layout Requirements for arm_convolve_wrapper_s8 API for IO and Learnables

nhkrishna9, 2025-01-30:
Hello,
I'm currently working on using the arm_convolve_wrapper_s8 function and need some guidance on how to correctly format the input data for this API. My application uses the NCHW data layout, where 'W' is the fastest-changing dimension. Consequently, my input, weight, bias, and output data are all structured in the NCHW format. However, I'm experiencing numerical discrepancies when I pass this format directly to the function.
Below is the code I'm using:
```
void custom_cmsisnn_conv2d_function(int8_T* output_buffer,
                            int8_T* input_buffer,
                            int8_T* weight_buffer,
                            int32_T* bias_buffer,
                            const int batch_size,
                            const int input_height,
                            const int input_width,
                            const int input_channels,
                            const int output_height,
                            const int output_width,
                            const int output_channels,
                            const int kernel_height,
                            const int kernel_width,
                            const int stride_height,
                            const int stride_width,
                            const int padding_height,
                            const int padding_width,
                            const int dilation_height,
                            const int dilation_width,
                            const int quant_multiplier,
                            const int quant_shift) {
 
  const cmsis_nn_dims in_dims = {.n = batch_size, .h = input_height, .w = input_width, .c = input_channels};
  const cmsis_nn_dims filt_dims = {.n = output_channels, .h = kernel_height, .w = kernel_width, .c = input_channels};
  const cmsis_nn_dims b_dims = {.n = 1, .h = 1, .w = 1, .c = output_channels};
  const cmsis_nn_dims out_dims = {.n = batch_size, .h = output_height, .w = output_width, .c = output_channels};
 
  int temp_buffer_size;
  int16_t *temp_buffer = NULL;
  temp_buffer_size = arm_convolve_s8_get_buffer_size(&in_dims, &filt_dims);
  if (temp_buffer_size > 0) {
    temp_buffer = (int16_t *)malloc((unsigned int)temp_buffer_size * sizeof(int16_t));
  }
 
  cmsis_nn_context context = {
      .buf = temp_buffer,
      .size = temp_buffer_size
  };
 
  cmsis_nn_conv_params conv_params;
  conv_params.input_offset  = 0;
  conv_params.output_offset = 0;
  conv_params.stride.h = stride_height;
  conv_params.stride.w = stride_width;
  conv_params.dilation.h = dilation_height;
  conv_params.dilation.w = dilation_width;
  conv_params.padding.h = padding_height;
  conv_params.padding.w = padding_width;
  conv_params.activation.min = -128;
  conv_params.activation.max = 127;
 
  cmsis_nn_per_channel_quant_params quant_params;
  int32_t* quant_multipliers = (int32_t*)malloc(output_channels * sizeof(int32_t));
  int32_t* quant_shifts = (int32_t*)malloc(output_channels * sizeof(int32_t));
  for (int i = 0; i < output_channels; ++i) {
       quant_multipliers[i] = quant_multiplier; // Example multiplier
       quant_shifts[i] = quant_shift;           // Example shift
  }
  quant_params.multiplier = quant_multipliers;
  quant_params.shift = quant_shifts;
 
  arm_convolve_wrapper_s8(&context, &conv_params, &quant_params,
&in_dims, input_buffer,
&filt_dims, weight_buffer,
&b_dims, bias_buffer,
&out_dims, output_buffer);
 
  if (temp_buffer_size > 0) {
      free(temp_buffer);
  }
  free(quant_multipliers);
  free(quant_shifts);
}

```
Could you please clarify the expected data formats for inputs, weights, biases, and outputs when using arm_convolve_wrapper_s8? Specifically, I would like to know if the function expects data in a format like HWCN (with 'N' being the fastest dimension) or another layout. Understanding this will help me ensure that the data is correctly aligned with the function's requirements.
Thank you for your assistance!

felix-johnny, 2025-01-30:
@nhkrishna9  
The data format expected is NHWC for the input activations and [out_ch, H, W, in_ch] for the weights..
Bias is a vector of output_channel number of entries. Output will be in NHWC format.

felix-johnny, 2025-01-31:
It is read right to left.. so, in_ch..

nhkrishna9, 2025-02-02:
Hi @felix-johnny,
Thank you for your reply. Are API users expected to write their own code and supply correctly formatted data to CMSIS-NN, or does CMSIS-NN offer any optimized APIs for data transformations?

felix-johnny, 2025-02-03:
@nhkrishna9  The API users are expected to supply the correctly formatted code to CMSIS-NN.

nhkrishna9, 2025-02-03:
@felix-johnny Thanks for your response.

closed 2025-02-03 by nhkrishna9