ARM-software/CMSIS-NN #170: arm_requantize_mve_pred with CMSIS_NN_USE_SINGLE_ROUNDING

cyril-enault, 2025-02-28:
arm_requantize_mve_pred (CMSIS_NN_USE_SINGLE_ROUNDING version) seems to have a different behaviour in rounding when compared to arm_requantize_mve (CMSIS_NN_USE_SINGLE_ROUNDING version)
regarding vqrdmulhq_m_n_s32 vs vqdmulhq_n_s32

   #ifdef CMSIS_NN_USE_SINGLE_ROUNDING
    const int right_shift = MIN(-1, shift);
    const int left_shift = shift - right_shift;
    const int32x4_t v_zero = vcreateq_s32(0, 0);

    const int32x4_t left_shift_dup = vdupq_x_n_s32(left_shift, p);
    const int32x4_t right_shift_dup = vdupq_x_n_s32(right_shift, p);

    int32x4_t result = **vqrdmulhq_m_n_s32**(v_zero, vshlq_m_s32(v_zero, val, left_shift_dup, p), multiplier, p);
    result = vrshlq_m_s32(v_zero, result, right_shift_dup, p);

    return result;
    #else
VS 
    const int right_shift = MIN(-1, shift);
    const int left_shift = shift - right_shift;

    const int32x4_t left_shift_dup = vdupq_n_s32(left_shift);
    const int32x4_t right_shift_dup = vdupq_n_s32(right_shift);

    int32x4_t result = **vqdmulhq_n_s32**(vshlq_s32(val, left_shift_dup), multiplier);
    result = vrshlq_s32(result, right_shift_dup);

    return result;

mansnils, 2025-03-05:
Hi @cyril-enault 
Yes it has different behavior which means it may produce different output. That is intentional. FYI, you will find the same flag in tensorflow.

closed 2025-03-05 by mansnils

cyril-enault, 2025-03-05:
Hi again,
I understood the difference with CMSIS_NN_USE_SINGLE_ROUNDING 
My point was the difference between
arm_requantize_mve_pred  and arm_requantize_mve 
Both with CMSIS_NN_USE_SINGLE_ROUNDING  flag.

While for arm_requantize_mve, vqdmulhq_n_s32 is used
for arm_requantize_mve_pred , vqrdmulhq_m_n_s32 is used (I would have expected vqdmulhq_m_n_s32 to be in line with vqdmulhq_n_s32).

mansnils, 2025-03-05:
Ah ok that might be an issue, reopening.

AdrianLundell, 2025-03-14:
Hi, the difference between `vqdmulhq_n_s32` and `vqrdmulhq_m_n_s32` is that  `vqrdmulhq_m_n_s32` is used for predication, which is why it is used for in the _pred variant of arm_requantize_mve. More info can be found here: https://developer.arm.com/architectures/instruction-sets/intrinsics/#q=vqrdmulhq

If you are experiencing numerical inaccuracies please come back to us with an example, otherwise I will close this issue again.