ARM-software/CMSIS-NN #177: Output diff when enabling CMSIS_NN_USE_REQUANTIZE_INLINE_ASSEMBLY

mansnils, 2025-05-02:
https://github.com/ARM-software/CMSIS-NN/pull/171 introduce optional flag CMSIS_NN_USE_REQUANTIZE_INLINE_ASSEMBLY. When this is enabled there are uplifts and as well as downlifts depending on the target and model. For one particular combination that is Arm Cortex-M7 using armclang there was an output diff observed for SSD Mobilenet v3.

mansnils, 2025-05-02:
This is hard to fix if you don't have access to the model and the compiler. We may be able to extract the "issue" from the model into a unit test to ease debugging.

MatthiasHertel80, 2025-05-06:
The MDK Community Edition license will allow full access to the model and the compiler. Since this is a non-commercial use-case, this is a perfect scenario to use it. 
Install and activate it either using an IDE of your choice: https://www.keil.arm.com/mdk-community/
Or in a GH CI environment as shown here: https://github.com/Arm-Examples/AVH_CI_Template/blob/main/.github/workflows/basic.yml
Or on the command line https://learn.arm.com/learning-paths/embedded-and-microcontrollers/vcpkg-tool-installation/licenseactivation/
This should equip anyone with the tools required to reproduce this.

ricardoasouz, 2026-10-10:
While investigating this I found that the inline assembly `arm_nn_divide_by_power_of_two()` returns a wrong result for a zero exponent (#247). It is not the cause of this diff, though: with the option enabled, `arm_nn_requantize()` and the other in-tree callers never pass a zero exponent.

ricardoasouz, 2026-10-19:
Correction to my earlier comment. The diff was reported on 7 March 2025, when the PR head was fef29eb0.

In that version, `arm_nn_requantize()` used `if (shift > 0)`, so a zero shift reached `arm_nn_divide_by_power_of_two()` with exponent 0, the case documented in #247. 

129dba39, the commit that added the flag and the README note about this diff, changed it to `shift >= 0`, so the merged code no longer takes that path.

The armclang Cortex-M7 disassembly attached to the PR on 31 March (`libcmsis-nn.a.disassembled.txt`) is that version: the divide path is entered for shift <= 0 (`cmp r2, #1` / `blt`). 

In 96 of its 200 divide sites, almost all in the per-channel convolution, depthwise and fully connected kernels, the exponent is computed with `negs` between the decrement and the `asrs`. From `arm_nn_vec_mat_mult_t_per_ch_s8`:

```
cmp.w   r6, #2147483648
it      hi
subhi   r6, #1
negs    r0, r2
asrs.w  r0, r6, r0
adc.w   r0, r0, #0
```

`negs` sets the carry when the shift is 0, `asrs` by 0 keeps it, and the `adc` adds 1, so every non-negative result of an output channel with shift 0 is off by one. The GCC disassembly of the same code has no such site.

A channel gets shift 0 when its effective scale is in [0.5, 1). If SSD MobileNet v3 has one, that would explain the diff, and the merged code should not show it.

Could you re-run it with the merged code to confirm?